martin-ger/esp32_nat_router #179: ESP32-S3

Leguq, 2024-11-30:
I bought an esp32-s3 just for this. I've been trying to get it work but everytime i flash the build, the windows disconnect sound keeps playing over and over again. The device ive been using is M5Dial, does anyone have a solution for this?

Leguq, 2024-12-01:
Hello,
I already fixed this issue using another person's build. Thanks for trying to help me out!

opcom-CMS, 2024-12-05:
> Hello, I already fixed this issue using another person's build. Thanks for trying to help me out!

@Leguq Which build did you use? I have an ESP32-S3 also and I would like to turn it into a NAT router as well.

Leguq, 2024-12-05:
> > Hello, I already fixed this issue using another person's build. Thanks for trying to help me out!
> 
> @Leguq Which build did you use? I have an ESP32-S3 also and I would like to turn it into a NAT router as well.
[esp32s3.zip](https://github.com/user-attachments/files/18022914/esp32s3.zip)

opcom-CMS, 2024-12-05:
> > > Hello, I already fixed this issue using another person's build. Thanks for trying to help me out!
> > 
> > 
> > @Leguq Which build did you use? I have an ESP32-S3 also and I would like to turn it into a NAT router as well.
> > [esp32s3.zip](https://github.com/user-attachments/files/18022914/esp32s3.zip)

@Leguq : Maybe a very basic question (I am sorry, I am new to that topic), but do you have a documentation, how I can upload this build to my ESP32-S3?

Leguq, 2024-12-06:
> @Leguq : Maybe a very basic question (I am sorry, I am new to that topic), but do you have a documentation, how I can upload this build to my ESP32-S3?

While i dont really have a documentation i can guide you, by the way my english is bad so if you have any issues reply to me here.
[toolinstall.txt](https://github.com/user-attachments/files/18034967/toolinstall.txt)

Leguq, 2024-12-06:
> @Leguq : Maybe a very basic question (I am sorry, I am new to that topic), but do you have a documentation, how I can upload this build to my ESP32-S3?

While i do not have a documentation i can guide you. My english is bad so if you have any issues please contact me by replying to me here.
[toolinstall.txt](https://github.com/user-attachments/files/18035068/toolinstall.txt)

Leguq, 2024-12-07:
> @Leguq : Maybe a very basic question (I am sorry, I am new to that topic), but do you have a documentation, how I can upload this build to my ESP32-S3?

[toolinstall.txt](https://github.com/user-attachments/files/18047830/toolinstall.txt)

opcom-CMS, 2024-12-12:
@Leguq Thank you!